uzh-rpg/rpg_svo #235: SVO crashes after a few seconds

akshatsh511, 2018-06-17:
Have been struggling with getting SVO to run on the provided data-set. Following is the back trace I get from gdb

`[ INFO] [1529271066.910056331]: Init: KLT 46.5785px average disparity.
[ INFO] [1529271066.943693181]: Init: KLT tracked 143 features
[ INFO] [1529271066.943767393]: Init: KLT 48.3242px average disparity.
[ INFO] [1529271066.975313537]: Init: KLT tracked 143 features
[ INFO] [1529271066.975392188]: Init: KLT 49.9558px average disparity.
[ INFO] [1529271067.008303615]: Init: KLT tracked 143 features
[ INFO] [1529271067.008384502]: Init: KLT 51.8553px average disparity.
[ INFO] [1529271067.009837697]: Init: Homography RANSAC 137 inliers.
[ INFO] [1529271067.010010750]: Init: Selected second frame, triangulated initial map.

Thread 7 "vo" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc2085700 (LWP 7812)]
0x00007ffff74cfb34 in svo::DepthFilter::initializeSeeds(boost::shared_ptr<svo::Frame>) () from /home/akshat/catkin_ws/devel/lib/libsvo.so
(gdb) bt
#0  0x00007ffff74cfb34 in svo::DepthFilter::initializeSeeds(boost::shared_ptr<svo::Frame>) () at /home/akshat/catkin_ws/devel/lib/libsvo.so
#1  0x00007ffff74d02a7 in svo::DepthFilter::updateSeedsLoop() () at /home/akshat/catkin_ws/devel/lib/libsvo.so
#2  0x00007ffff330bbcd in  () at /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#3  0x00007ffff3c686db in start_thread (arg=0x7fffc2085700) at pthread_create.c:463
#4  0x00007ffff595688f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
`

Configuration I have been running is 
ROS melodic
Ubuntu 18.04 - 64 bit ( although similar issues faced with 14.04)

Please help since I have been stuck on this for a long time now

WinKILLER, 2018-10-04:
I tried both of methods recommended by yan99033 on ubuntu 18.04-64 with ROS Melodic. I experimented with live camera and a bag file. Unfortunately issue is not solved - still having error at 
(0x00007ffff74acd44 in svo::DepthFilter::initializeSeeds(boost::shared_ptr<svo::Frame>)

gkouros, 2019-01-14:
I am facing the exact same issue using the latest version of svo with ros melodic. Did anyone find any solutions?

gkouros, 2019-01-14:
After some investigation, I found that the crash occurs at line 125 of rpg_svo/svo/src/depth_filter.cpp. Any ideas why that happens?

gkouros, 2019-01-16:
It is fixed by pull request #238
https://github.com/uzh-rpg/rpg_svo/pull/238

AndreiBaraian, 2019-03-21:
I have done the modifications in pull request #238, and now it works. But the log stops after [ INFO] [1553165839.367933763]: Init: Selected second frame, triangulated initial map. The rest of the algorithm seems to be working fine in RVIZ, but I don't understand why the log stops. Did any of you encountered the same problem? I also run ubuntu 18.04 with ROS melodic.

Ador2, 2019-06-18:
@AndreiBaraian I met the exact same issue, done the modifications too, no luck.