uzh-rpg/rpg_svo #236: Depth_filter crashes on Odroid UX4 and MacOS?

weiminshen99, 2018-06-20:
Dear Everyone,
I am new for SVO and could be completely wrong, but the thread "depth_filter" does not seem working properly on both Odroid and MacOS, and crashes on the given dataset as follows. Please tell me how to do things correctly. Thanks!

===== On Odroid UX4 =====
odroid@odroid:~/GitHub/SVO/rpg_svo/svo$ ./bin/test_depth_filter
RUN EXPERIMENT: read 187 dataset entries.
reading image: '/home/odroid/GitHub/SVO/Datasets/sin2_tex2_h1_v8_d/img/frame_000002_0.png'
[INFO] DepthFilter: Initialized 570 new seeds
reading image: '/home/odroid/GitHub/SVO/Datasets/sin2_tex2_h1_v8_d/img/frame_000003_0.png'
reading image: '/home/odroid/GitHub/SVO/Datasets/sin2_tex2_h1_v8_d/img/frame_000004_0.png'
test_depth_filter: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:109: 
Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() 
[with T = double; int Size = 36; int MatrixOrArrayOptions = 0]: Assertion 
`(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (15)) == 0
&& "this assertion is explained here:
" "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" 
" **** READ THIS WEB PAGE !!! ****"' failed.
Aborted

odroid@odroid:~/GitHub/SVO/rpg_svo/svo$ ./bin/test_pipeline
[INFO] SVO initialized
[INFO] DepthFilter starting thread
reading image /home/odroid/GitHub/SVO/Datasets/sin2_tex2_h1_v8_d/img/frame_000002_0.png
[INFO] RESET
[INFO] Init: Selected first frame.
Frame-Id: 0     #Features: 0    Proc. Time: 41.441ms
[INFO] Init: KLT tracked 565 features
[INFO] Init: KLT 25.3823px average disparity.
Frame-Id: 1     #Features: 0    Proc. Time: 64.492ms
[INFO] Init: KLT tracked 543 features
[INFO] Init: KLT 50.8035px average disparity.
[INFO] Init: Homography RANSAC 543 inliers.
[INFO] Init: Selected second frame, triangulated initial map.
Frame-Id: 2     #Features: 497  Proc. Time: 78.409ms
Frame-Id: 3     #Features: 121  Proc. Time: 17.184ms
Frame-Id: 4     #Features: 121  Proc. Time: 7.487ms
Frame-Id: 5     #Features: 121  Proc. Time: 6.605ms
Frame-Id: 6     #Features: 121  Proc. Time: 23.065ms
Frame-Id: 7     #Features: 121  Proc. Time: 7.286ms
..........................................
Frame-Id: 182   #Features: 121  Proc. Time: 9.914ms
Frame-Id: 183   #Features: 121  Proc. Time: 8.66ms
Frame-Id: 184   #Features: 121  Proc. Time: 11.243ms
Frame-Id: 185   #Features: 121  Proc. Time: 8.694ms
[INFO] DepthFilter stop thread invoked.
[INFO] DepthFilter interrupt and join thread...
[INFO] DepthFilter destructed.
[INFO] SVO destructor invoked
[INFO] Map destructed
BenchmarkNode finished.
odroid@odroid:~/GitHub/SVO/rpg_svo/svo$

===== On MacOS =======
The above errors happened in a similar fashion.

PLEASE HELP!!!

WANGLXUE, 2019-06-16:
I meet the same problem.
How did you solve it?

besttangent, 2020-06-01:
Is there anyone who can solve it ?

lenardxu, 2022-06-20:
This problem is actually, as pointed out in the built-in link in the error message, a eigen alignment problem. This kind of alignment is designated for sse accleration / optimization. I suppose the problem is triggered by your machine, which should be 32-bit. Anyway, here is a compromise. You can disable all related functionalities to circumvent this problem, but at cost of efficiency for your information.
As I found, to launch ros node `svo_node`, two CMake files are concerned, i.e., ".../svo_ws/src/rpg_svo_pro_open/svo_ros/CMakeLists.txt" and ".../svo_ws/src/rpg_svo_pro_open/svo_cmake/cmake/Modules/SvoSetup.cmake", the latter of which is included in the former one. For purpose of deacticating the alignment functionality of eigen, you need to add two definitions in both these CMake files:
```
add_definitions(-DEIGEN_DONT_VECTORIZE)
add_definitions(-DEIGEN_DISABLE_UNALIGNED_ARRAY_ASSERT)
```