uzh-rpg/rpg_svo #237: Sophus cant build

PsouthRobotics1182, 2018-07-21:
If i checkout Sophus commit a621ff, and try to build with make i get this error

```

`usr/bin/cmake -H/home/drew/svo/workspace/Sophus -B/home/drew/svo/workspace/Sophus/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/drew/svo/workspace/Sophus/build/CMakeFiles /home/drew/svo/workspace/Sophus/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/drew/svo/workspace/Sophus/build'
make -f CMakeFiles/Sophus.dir/build.make CMakeFiles/Sophus.dir/depend
make[2]: Entering directory '/home/drew/svo/workspace/Sophus/build'
cd /home/drew/svo/workspace/Sophus/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/drew/svo/workspace/Sophus /home/drew/svo/workspace/Sophus /home/drew/svo/workspace/Sophus/build /home/drew/svo/workspace/Sophus/build /home/drew/svo/workspace/Sophus/build/CMakeFiles/Sophus.dir/DependInfo.cmake --color=
Scanning dependencies of target Sophus
make[2]: Leaving directory '/home/drew/svo/workspace/Sophus/build'
make -f CMakeFiles/Sophus.dir/build.make CMakeFiles/Sophus.dir/build
make[2]: Entering directory '/home/drew/svo/workspace/Sophus/build'
[  5%] Building CXX object CMakeFiles/Sophus.dir/sophus/so2.cpp.o
/usr/bin/c++   -DSophus_EXPORTS -I/usr/include/eigen3  -O3 -DNDEBUG  -fPIC   -Wall -Werror -Wno-unused-variable                    -Wno-unused-but-set-variable -Wno-unknown-pragmas  -o CMakeFiles/Sophus.dir/sophus/so2.cpp.o -c /home/drew/svo/workspace/Sophus/sophus/so2.cpp
/home/drew/svo/workspace/Sophus/sophus/so2.cpp: In constructor ‘Sophus::SO2::SO2()’:
/home/drew/svo/workspace/Sophus/sophus/so2.cpp:32:26: error: lvalue required as left operand of assignment
   unit_complex_.real() = 1.;
                          ^~
/home/drew/svo/workspace/Sophus/sophus/so2.cpp:33:26: error: lvalue required as left operand of assignment
   unit_complex_.imag() = 0.;
                          ^~
CMakeFiles/Sophus.dir/build.make:65: recipe for target 'CMakeFiles/Sophus.dir/sophus/so2.cpp.o' failed
make[2]: *** [CMakeFiles/Sophus.dir/sophus/so2.cpp.o] Error 1
make[2]: Leaving directory '/home/drew/svo/workspace/Sophus/build'
CMakeFiles/Makefile2:107: recipe for target 'CMakeFiles/Sophus.dir/all' failed
make[1]: *** [CMakeFiles/Sophus.dir/all] Error 2
make[1]: Leaving directory '/home/drew/svo/workspace/Sophus/build'
Makefile:141: recipe for target 'all' failed
make: *** [all] Error 2
```
`
and if I use a later version of sophus which doesnt have this problem i get this error when using 'catkin make' at the end to build rpg-svo

```

'Base path: /home/drew/svo/catkin_ws
Source space: /home/drew/svo/catkin_ws/src
Build space: /home/drew/svo/catkin_ws/build
Devel space: /home/drew/svo/catkin_ws/devel
Install space: /home/drew/svo/catkin_ws/install
####
#### Running command: "cmake /home/drew/svo/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/drew/svo/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/drew/svo/catkin_ws/install -G Unix Makefiles" in "/home/drew/svo/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/drew/svo/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/drew/svo/svo_install_overlay_ws/devel;/opt/ros/melodic;/home/drew/svo/svo_install_ws/install
-- This workspace overlays: /home/drew/svo/svo_install_overlay_ws/devel;/opt/ros/melodic;/home/drew/svo/svo_install_ws/install
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.13", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/drew/svo/catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.13") 
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 8 packages in topological order:
-- ~~  - rqt_svo
-- ~~  - svo_analysis
-- ~~  - svo_msgs
-- ~~  - vikit_common
-- ~~  - vikit_py
-- ~~  - vikit_ros
[... 180 lines cut ...]
/home/drew/svo/catkin_ws/src/rpg_vikit/vikit_common/include/vikit/math_utils.h:14:24: fatal error: sophus/se3.h: No such file or directory
 #include <sophus/se3.h>
                        ^
compilation terminated.
rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/build.make:62: recipe for target 'rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/atan_camera.cpp.o' failed
make[2]: *** [rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/atan_camera.cpp.o] Error 1
In file included from /home/drew/svo/catkin_ws/src/rpg_vikit/vikit_common/src/robust_cost.cpp:24:0:
/home/drew/svo/catkin_ws/src/rpg_vikit/vikit_common/include/vikit/math_utils.h:14:24: fatal error: sophus/se3.h: No such file or directory
 #include <sophus/se3.h>
                        ^
compilation terminated.
rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/build.make:182: recipe for target 'rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/robust_cost.cpp.o' failed
make[2]: *** [rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/robust_cost.cpp.o] Error 1
[ 36%] Built target svo_msgs_generate_messages_eus
Scanning dependencies of target svo_msgs_generate_messages
[ 36%] Built target svo_msgs_generate_messages
CMakeFiles/Makefile2:1666: recipe for target 'rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/all' failed
make[1]: *** [rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed'

```

it clearly cant find sophus but I cant figure out why

ahmadyan, 2018-10-02:
It is a known issue with that commit. Try replacing that line with

cesarhcq, 2019-01-15:
> It is a known issue with that commit. Try replacing that line with
> 
> `unit_complex_ = std::complex<double>(1,0);`

That's fix my problem!

Thank you!

niharricky27, 2019-04-23:
> It is a known issue with that commit. Try replacing that line with
> 
> `unit_complex_ = std::complex<double>(1,0);`

After replacing that line in the mentioned commit, and doing catkin_make on catkin_ws as advised in [rpg_svo installation instructions](https://github.com/uzh-rpg/rpg_svo/wiki/Installation:-ROS) 

I'm getting this error:
`~/catkin_ws$ catkin_make
Base path: /home/user/catkin_ws
Source space: /home/user/catkin_ws/src
Build space: /home/user/catkin_ws/build
Devel space: /home/user/catkin_ws/devel
Install space: /home/user/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/user/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/user/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/user/rosjava/devel;/opt/ros/melodic
-- This workspace overlays: /home/user/rosjava/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.15", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/user/catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.15") 
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.17
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 9 packages in topological order:
-- ~~  - beginner_tutorials
-- ~~  - rqt_svo
-- ~~  - svo_analysis
-- ~~  - svo_msgs
-- ~~  - vikit_common
-- ~~  - vikit_py
-- ~~  - vikit_ros
-- ~~  - svo
-- ~~  - svo_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'beginner_tutorials'
-- ==> add_subdirectory(beginner_tutorials)
-- +++ processing catkin package: 'rqt_svo'
-- ==> add_subdirectory(rpg_svo/rqt_svo)
-- +++ processing catkin package: 'svo_analysis'
-- ==> add_subdirectory(rpg_svo/svo_analysis)
-- +++ processing catkin package: 'svo_msgs'
-- ==> add_subdirectory(rpg_svo/svo_msgs)
-- Using these message generators: gencpp;geneus;genjava;genlisp;gennodejs;genpy
-- svo_msgs: 4 messages, 0 services
-- +++ processing catkin package: 'vikit_common'
-- ==> add_subdirectory(rpg_vikit/vikit_common)
-- Eigen found (include: /usr/include/eigen3)
-- +++ processing catkin package: 'vikit_py'
-- ==> add_subdirectory(rpg_vikit/vikit_py)
-- +++ processing catkin package: 'vikit_ros'
-- ==> add_subdirectory(rpg_vikit/vikit_ros)
-- Using these message generators: gencpp;geneus;genjava;genlisp;gennodejs;genpy
CMake Warning at /opt/ros/melodic/share/cmake_modules/cmake/Modules/FindEigen.cmake:62 (message):
  The FindEigen.cmake Module in the cmake_modules package is deprecated.

  Please use the FindEigen3.cmake Module provided with Eigen.  Change
  instances of find_package(Eigen) to find_package(Eigen3).  Check the
  FindEigen3.cmake Module for the resulting CMake variable names.

[... 92 lines cut ...]
/home/user/catkin_ws/src/rpg_vikit/vikit_common/src/img_align.cpp: In member function ‘virtual void vk::SecondOrderMinimisationSE3::finishIteration()’:
/home/user/catkin_ws/src/rpg_vikit/vikit_common/src/img_align.cpp:437:34: error: ‘CV_WINDOW_AUTOSIZE’ was not declared in this scope
     cv::namedWindow("residuals", CV_WINDOW_AUTOSIZE);
                                  ^~~~~~~~~~~~~~~~~~
/home/user/catkin_ws/src/rpg_vikit/vikit_common/src/img_align.cpp:437:34: note: suggested alternative: ‘CV_MINOR_VERSION’
     cv::namedWindow("residuals", CV_WINDOW_AUTOSIZE);
                                  ^~~~~~~~~~~~~~~~~~
                                  CV_MINOR_VERSION
rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/build.make:230: recipe for target 'rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/pinhole_camera.cpp.o' failed
make[2]: *** [rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/pinhole_camera.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/build.make:278: recipe for target 'rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/img_align.cpp.o' failed
make[2]: *** [rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/img_align.cpp.o] Error 1
rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/build.make:254: recipe for target 'rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/homography.cpp.o' failed
make[2]: *** [rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/homography.cpp.o] Error 1
CMakeFiles/Makefile2:2033: recipe for target 'rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/all' failed
make[1]: *** [rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`
Am I missing anything?
Please do let me know if I should give any other type of details or should edit my query someway.

Thanks.

xiaoxiao5555, 2024-08-09:
unit_complex_.real(1.);
unit_complex_.imag(0.);

tdcsu, 2025-03-06:
> unit_complex_.real(1.); unit_complex_.imag(0.);
thanks, it works well