uzh-rpg/rpg_svo #242: Can Raspberry Pi run SVO?

WR0903, 2018-10-24:
Can Raspberry Pi run SVO? How to configure it.

odinhr, 2018-10-30:
Hi @WR0903 .

I've tried it on Raspberry Pi and it works aprox. at 6 Hz. Everytime it was lost and cannot support fast movements. If your problems are compiling it, maybe you should try telling the compiler you are ARM .  I don't recommend you to run on RPI. Try with Odroid or something similar.

Good luck.

WR0903, 2018-10-31:
Thank you very much for your guidance. I compiled it a few days ago and it runs faster than I thought.

deepak-1530, 2019-03-30:
Hey how to change the parameters for the raspi cam? Its not able to detect features

baranovilyaserg, 2021-09-02:
Hi, how did you end up installing SVO on raspberry? Can you write the instructions?