uzh-rpg/rpg_svo #245: error ncamera

FPSychotic, 2019-01-31:
>  * /svo/
> 
> PARAMETERS
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.14
>  * /svo/calib_file: ~/svo_install_ove...
>  * /svo/cam0_topic: /camera/rgb/image...
>  * /svo/kfselect_criterion: FORWARD
>  * /svo/kfselect_min_angle: 10
>  * /svo/kfselect_min_disparity: 40
>  * /svo/kfselect_min_dist_metric: 0.01
>  * /svo/kfselect_numkfs_lower_thresh: 70
>  * /svo/kfselect_numkfs_upper_thresh: 110
>  * /svo/map_scale: 1.0
>  * /svo/max_fts: 120
>  * /svo/quality_max_drop_fts: 40
>  * /svo/quality_min_fts: 70
>  * /svo/relocalization_max_trials: 100
>  * /svo/update_seeds_with_old_keyframes: True
> 
> NODES
>   /
>     svo (svo_ros/svo_node)
>     svo_gui (rqt_gui/rqt_gui)
>     vis (rviz/rviz)
> 
> ROS_MASTER_URI=http://localhost:11311
> 
> process[svo-1]: started with pid [10156]
> process[vis-2]: started with pid [10157]
> process[svo_gui-3]: started with pid [10158]
> [ WARN] [1548942300.708201706]: Cannot find value for parameter: pipeline_is_stereo, assigning default: 0
> [ WARN] [1548942300.712570621]: Cannot find value for parameter: set_initial_attitude_from_gravity, assigning default: 1
> [ WARN] [1548942300.717867872]: Cannot find value for parameter: automatic_reinitialization, assigning default: 0
> [ INFO] [1548942300.726973036]: Found parameter: calib_file, value: ~/svo_install_overlay_ws/src/rpg_svo_example/svo_ros/calib/r200.yaml
> E0131 13:45:00.727658 10156 ncamera.cpp:54] Failed to load NCamera from file ~/svo_install_overlay_ws/src/rpg_svo_example/svo_ros/calib/r200.yaml with the error: 
> yaml-cpp: error at line 0, column 0: bad file
> *** Aborted at 1548942300 (unix time) try "date -d @1548942300" if you are using GNU date ***
> PC: @                0x0 (unknown)
> *** SIGSEGV (@0x20) received by PID 10156 (TID 0x7f974a783ec0) from PID 32; stack trace: ***
>     @     0x7f97496bf4b0 (unknown)
>     @     0x7f974a03c455 svo::factory::loadCameraFromYaml()
>     @     0x7f974a03e09c svo::factory::makeMono()
>     @     0x7f974a01ad02 svo::SvoInterface::SvoInterface()
>     @     0x7f974a0412b1 svo_ros::SvoNodeBase::SvoNodeBase()
>     @           0x400c58 main
>     @     0x7f97496aa830 __libc_start_main
>     @           0x400ce9 _start
>     @                0x0 (unknown)
> [svo-1] process has died [pid 10156, exit code -11, cmd /home/ihexa/svo_install_overlay_ws/devel/lib/svo_ros/svo_node __name:=svo __log:=/home/ihexa/.ros/log/1df5371a-254f-11e9-876a-e84e0667c52b/svo-1.log].
> log file: /home/ihexa/.ros/log/1df5371a-254f-11e9-876a-e84e0667c52b/svo-1*.log

------------------------------------------------------------------------------------------------------------

i dont know what mean that error i tried everything
I looking everywere to now what mean ncamera.
i tested a lot of different configs from examples etc allways the same error.

kyuminKim00, 2023-09-19:
I have the same problem as you.
Did you solve the problem?

FPSychotic, 2023-09-19:
This is obsolete and there is better solutions.  Also check by SVO Pro, which is a newer version