uzh-rpg/rpg_svo #246: non-ROS flavor test_pipeline exception on Mac

yf13, 2019-03-31:
I built the test_pipeline with llvm 7.0 on Mac, then play with sample data "sin2_tex2_h1_v8_d". I am using the Release with debug info build type thus local BA not enabled.

Sometimes (not often) test_pipeline finishes w/o any exception, but most time it finishes with an error at the cleanup stage:

> Frame-Id: 185 	#Features: 121 	Proc. Time: 1.915ms 
last position: 11.539 4.90906 -0.00978625
 [INFO] DepthFilter stop thread invoked.
 [INFO] DepthFilter interrupt and join thread... 
 [INFO] DepthFilter destructed.
test_pipeline(6773,0x7fffc47a33c0) malloc: *** error for object 0x3e21d75abf37abac: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Abort trap: 6

Then I tried to use lldb get the back trace:

> (lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fffbb9bbd42 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fffbbaa9457 libsystem_pthread.dylib`pthread_kill + 90
    frame #2: 0x00007fffbb921420 libsystem_c.dylib`abort + 129
    frame #3: 0x00007fffbba10fe7 libsystem_malloc.dylib`free + 530
    frame #4: 0x00000001000bf1d5 libsvo.dylib`svo::initialization::KltHomographyInit::~KltHomographyInit() [inlined] Eigen::internal::handmade_aligned_free(ptr=<unavailable>) at Memory.h:98 [opt]
    frame #5: 0x00000001000bf1cc libsvo.dylib`svo::initialization::KltHomographyInit::~KltHomographyInit() [inlined] Eigen::internal::aligned_free(ptr=<unavailable>) at Memory.h:179 [opt]
    frame #6: 0x00000001000bf1cc libsvo.dylib`svo::initialization::KltHomographyInit::~KltHomographyInit() [inlined] Eigen::aligned_allocator<Eigen::Matrix<double, 3, 1, 0, 3, 1> >::deallocate(p=<unavailable>) at Memory.h:763 [opt]
    frame #7: 0x00000001000bf1cc libsvo.dylib`svo::initialization::KltHomographyInit::~KltHomographyInit() [inlined] std::__1::allocator_traits<Eigen::aligned_allocator<Eigen::Matrix<double, 3, 1, 0, 3, 1> > >::deallocate(__p=<unavailable>) at memory:1556 [opt]
    frame #8: 0x00000001000bf1cc libsvo.dylib`svo::initialization::KltHomographyInit::~KltHomographyInit() [inlined] std::__1::__vector_base<Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 3, 1, 0, 3, 1> > >::~__vector_base(this=<unavailable>) at vector:459 [opt]
    frame #9: 0x00000001000bf1b9 libsvo.dylib`svo::initialization::KltHomographyInit::~KltHomographyInit() [inlined] std::__1::vector<Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 3, 1, 0, 3, 1> > >::~vector(this=<unavailable>) at iterator:1427 [opt]
    frame #10: 0x00000001000bf1b9 libsvo.dylib`svo::initialization::KltHomographyInit::~KltHomographyInit() [inlined] std::__1::vector<Eigen::Matrix<double, 3, 1, 0, 3, 1>, std::__1::allocator<Eigen::Matrix<double, 3, 1, 0, 3, 1> > >::~vector(this=<unavailable>) at global.h:36 [opt]
    frame #11: 0x00000001000bf1b9 libsvo.dylib`svo::initialization::KltHomographyInit::~KltHomographyInit() [inlined] std::__1::vector<Eigen::Matrix<double, 3, 1, 0, 3, 1>, std::__1::allocator<Eigen::Matrix<double, 3, 1, 0, 3, 1> > >::~vector(this=<unavailable>) at global.h:36 [opt]
    frame #12: 0x00000001000bf1b9 libsvo.dylib`svo::initialization::KltHomographyInit::~KltHomographyInit(this=0x0000000116000380) at initialization.h:39 [opt]
    frame #13: 0x00000001000bb796 libsvo.dylib`svo::FrameHandlerMono::~FrameHandlerMono() [inlined] svo::initialization::KltHomographyInit::~KltHomographyInit(this=<unavailable>) at initialization.h:39 [opt]
    frame #14: 0x00000001000bb791 libsvo.dylib`svo::FrameHandlerMono::~FrameHandlerMono(this=0x0000000116000020) at frame_handler_mono.cpp:57 [opt]
    frame #15: 0x00000001000bb8be libsvo.dylib`svo::FrameHandlerMono::~FrameHandlerMono() [inlined] svo::FrameHandlerMono::~FrameHandlerMono(this=0x0000000116000020) at frame_handler_mono.cpp:55 [opt]
    frame #16: 0x00000001000bb8b9 libsvo.dylib`svo::FrameHandlerMono::~FrameHandlerMono(this=0x0000000116000020) at frame_handler_mono.cpp:55 [opt]
    frame #17: 0x000000010000f67c test_pipeline`main [inlined] svo::BenchmarkNode::~BenchmarkNode() at test_pipeline.cpp:56 [opt]
    frame #18: 0x000000010000f66d test_pipeline`main [inlined] svo::BenchmarkNode::~BenchmarkNode() at test_pipeline.cpp:55 [opt]
    frame #19: 0x000000010000f66d test_pipeline`main(argc=<unavailable>, argv=<unavailable>) at test_pipeline.cpp:103 [opt]
    frame #20: 0x00007fffbb88d235 libdyld.dylib`start + 1

My Eigen library is 3.3.7. 

Not sure if anyone has met similar issue and got a fix?

WANGLXUE, 2019-06-17:
Same problem.
So how did you solve it?

wongfei, 2020-01-22:
for all those who crashing inside ~KltHomographyInit

modify files:
rpg_vikit/vikit_common/include/vikit/math_utils.h
rpg_vikit/vikit_common/include/vikit/homography.h

```
#include <Eigen/Core>
#include <Eigen/StdVector>
// THIS IS SPARTA!!! DIE IN HELL WHILE DEBUGGING!!!
#ifndef RPG_SVO_VIKIT_IS_VECTOR_SPECIALIZED
#define RPG_SVO_VIKIT_IS_VECTOR_SPECIALIZED
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Eigen::Vector3d)
EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION(Eigen::Vector2d)
#endif
```