uzh-rpg/rpg_svo #247: Running SVO on a live camera stream

hassanyf, 2019-04-15:
I can run SVO fine from the dataset but how do I run a live camera stream? I have a PS3 Eye webcam and I don't know how to create a ros node that publishes images on the topic `camera/image`. Is there any tutorial on it?

ouyangandy, 2019-04-29:
Is it necessary to convert a color image to a grayscale image when publishing a node, or is it already done this step?

canberkgurel, 2019-06-21:
Install USB_CAM:

```
cd ~catkin_ws/src 

git clone https://github.com/bosch-ros-pkg/usb_cam.git 

cd ~/catkin_ws

catkin_make

source devel/setup.bash

gedit ~/catkin_ws/devel/lib/usb_cam/usb_cam.launch
```

Add the following and save

```
<launch>
  <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video0" />
    <param name="image_width" value="640" />
    <param name="image_height" value="480" />
    <param name="pixel_format" value="mjpeg" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="io_method" value="mmap"/>
  </node>
  <node name="image_view" pkg="image_view" type="image_view" respawn="false" output="screen">
    <remap from="image" to="/usb_cam/image_raw"/>
    <remap from="/usb_cam/image_raw" to="/camera/image"/>
    <param name="autosize" value="true" />
  </node>
</launch>

roslaunch usb_cam usb_cam.launch
```

closed 2019-06-24 by hassanyf