uzh-rpg/rpg_svo #251: Not working on Melodic + Ubuntu 18.04

wertig0n, 2019-06-09:
Tried to follow installation guide, but Sophus isn't cooperating. When I checkout a621ff and then subsequently try to make, I get:

`[  5%] Building CXX object CMakeFiles/Sophus.dir/sophus/so2.cpp.o
/home/per/workspace/Sophus/sophus/so2.cpp: In constructor ‘Sophus::SO2::SO2()’:
/home/per/workspace/Sophus/sophus/so2.cpp:32:26: error: lvalue required as left operand of assignment
   unit_complex_.real() = 1.;
                          ^~
/home/per/workspace/Sophus/sophus/so2.cpp:33:26: error: lvalue required as left operand of assignment
   unit_complex_.imag() = 0.;
                          ^~
CMakeFiles/Sophus.dir/build.make:65: recipe for target 'CMakeFiles/Sophus.dir/sophus/so2.cpp.o' failed
make[2]: *** [CMakeFiles/Sophus.dir/sophus/so2.cpp.o] Error 1
CMakeFiles/Makefile2:144: recipe for target 'CMakeFiles/Sophus.dir/all' failed
make[1]: *** [CMakeFiles/Sophus.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
`

Trying to build from master is a no-go as well. Relevant information:

`$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic`

`$ dpkg -l ros-melodic-desktop-full
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  ros-melodic-de 1.4.1-0bioni amd64        A metapackage to aggregate severa`

`$ g++ --version
g++ (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.`

Would appreciate any help available.

Ador2, 2019-06-12:
that Sophus commit is still not fixed.
see https://github.com/uzh-rpg/rpg_svo/issues/237#issuecomment-426409963

guin18, 2020-11-21:
try solution in #237? I managed to fix it.