uzh-rpg/rpg_svo #254: Help creating environment - feature detection

indajm, 2019-07-25:
I would like to create a nice environment so the SVO package works well. I'm using an AR.Drone 2.0. The bottom camera is a QVGA (320*240) and works at 20 Hz. The environment where this package is going to be used is a 80cm x 80cm platform and nothing outside it. In the platform I can put all the images or things that I want. 

This is how the platform looks now (the white papers with some AR tags on them), and SVO says that there are not enough features:
[image]

So I added some black and white squares between some tags and it seems to work:

[image]

but then I move the camera a little and this is what happens:

[image]

So, how do you recommend me to design that platform? Maybe adding a lot of black and white squares between all tags? Or is there anything better to do?

Thanks

WANGLXUE, 2019-09-11:
I think you should adjust some parameters in Config.cpp. I also failed in feature tracking. For different size and information of images, parameters should be changed first for this SVO. Maybe it will help, but I am not sure.

WANGLXUE, 2019-09-11:
Sorry for not attaching the links.
https://github.com/uzh-rpg/rpg_svo/issues/205