uzh-rpg/rpg_svo #259: Can I run svo on my own bag which only has image frames in topic?

Lucksong, 2019-11-12:
As the log shows: My bag only has /dvs/image_raw, and the example bag has many other topics, what should I do to run svo on my own bag?

**usr@usr:~/svo_ws/src/SVO/rpg_svo/svo_ros/data$ rosbag info airground_rig_s3_2013-03-18_21-38-48.bag** 
path:        airground_rig_s3_2013-03-18_21-38-48.bag
version:     2.0
duration:    2:42s (162s)
start:       Mar 19 2013 04:38:50.32 (1363639130.32)
end:         Mar 19 2013 04:41:32.76 (1363639292.76)
size:        1.7 GB
messages:    61273
compression: none [1625/1625 chunks]
types:       geometry_msgs/PoseStamped    [d3812c3cbc69362b77dc0b19b345f8f5]
             geometry_msgs/Vector3Stamped [7b324c7325e683bf02a9b14b01090ec7]
             sensor_msgs/CameraInfo       [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/Image            [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu              [6a62c6daae103f4ff57a132d6f95cec2]
topics:      /imu/magnetometer    16244 msgs    : geometry_msgs/Vector3Stamped
             Rig                  19039 msgs    : geometry_msgs/PoseStamped   
             camera/camera_info    4873 msgs    : sensor_msgs/CameraInfo      
             camera/image_raw      4873 msgs    : sensor_msgs/Image           
             imu/data             16244 msgs    : sensor_msgs/Imu
**usr@usr:~/svo_ws/src/SVO/rpg_svo/svo_ros/data$ rosbag info DAVIS240C-2018-11-26T16-51-05+0800-08360021-0.bag** 
path:        DAVIS240C-2018-11-26T16-51-05+0800-08360021-0.bag
version:     2.0
duration:    15.5s
start:       Jan 01 1970 08:03:43.94 (223.94)
end:         Jan 01 1970 08:03:59.39 (239.39)
size:        115.6 MB
messages:    533
compression: none [126/126 chunks]
types:       dvs_msgs/EventArray [5e8beee5a6c107e504c2e78903c224b8]
             sensor_msgs/Image   [060021388200f6f0f447d0fcd9c64743]
topics:      /dvs/events      349 msgs    : dvs_msgs/EventArray
             /dvs/image_raw   184 msgs    : sensor_msgs/Image

WANGLXUE, 2019-11-21:
Of course it works with only image topic in bag file.
But the topic published should be camera/image_raw.

zhaodejian, 2021-04-11:
I also want to run SVO with my own rosbag, but I don't even have an image, as if nothing happened @Lucksong