uzh-rpg/rpg_svo #262: [svo-1] process has died [pid 2709, exit code -11

chinthaA, 2020-02-12:
I'm new to ROS and trying to run the SVO and ended up with following error. Im using ros melodic in ubuntu 18. I trying to run for the esample data set provided. Please help me to correct it

hinthaka@chinthaka-VirtualBox:~$ roslaunch svo_ros test_rig3.launch
... logging to /home/chinthaka/.ros/log/06d84934-4d4c-11ea-8360-08002776c300/roslaunch-chinthaka-VirtualBox-2693.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://chinthaka-VirtualBox:35921/

SUMMARY
========

CLEAR PARAMETERS
 * /svo/

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /svo/cam_cx: 0.45905
 * /svo/cam_cy: 0.510056
 * /svo/cam_d0: 0.932
 * /svo/cam_fx: 0.509326
 * /svo/cam_fy: 0.796651
 * /svo/cam_height: 480
 * /svo/cam_model: ATAN
 * /svo/cam_topic: /camera/image_raw
 * /svo/cam_width: 752
 * /svo/grid_size: 30
 * /svo/init_rx: 3.14
 * /svo/init_ry: 0.0
 * /svo/init_rz: 0.0
 * /svo/loba_num_iter: 0
 * /svo/max_n_kfs: 10

NODES
  /
    svo (svo_ros/vo)

ROS_MASTER_URI=http://localhost:11311

process[svo-1]: started with pid [2709]
create vo_node
[ WARN] [1581480028.515134340]: Cannot find value for parameter: svo/publish_img_pyr_level, assigning default: 0
[ WARN] [1581480028.517829986]: Cannot find value for parameter: svo/publish_every_nth_img, assigning default: 1
[ WARN] [1581480028.518570619]: Cannot find value for parameter: svo/publish_every_nth_dense_input, assigning default: 1
[ WARN] [1581480028.521074302]: Cannot find value for parameter: svo/publish_world_in_cam_frame, assigning default: 1
[ WARN] [1581480028.522286588]: Cannot find value for parameter: svo/publish_map_every_frame, assigning default: 0
[ WARN] [1581480028.523074612]: Cannot find value for parameter: svo/publish_point_display_time, assigning default: 0
[ WARN] [1581480028.688583358]: Cannot find value for parameter: svo/publish_markers, assigning default: 1
[ WARN] [1581480028.689124586]: Cannot find value for parameter: svo/publish_dense_input, assigning default: 0
[ WARN] [1581480028.689539140]: Cannot find value for parameter: svo/accept_console_user_input, assigning default: 1
[ INFO] [1581480028.690255192]: Found parameter: svo/cam_model, value: ATAN
[ INFO] [1581480028.690798839]: Found parameter: svo/cam_width, value: 752
[ INFO] [1581480028.691272774]: Found parameter: svo/cam_height, value: 480
[ INFO] [1581480028.691703212]: Found parameter: svo/cam_fx, value: 0.509326
[ INFO] [1581480028.692213719]: Found parameter: svo/cam_fy, value: 0.796651
[ INFO] [1581480028.692622041]: Found parameter: svo/cam_cx, value: 0.45905
[ INFO] [1581480028.693089482]: Found parameter: svo/cam_cy, value: 0.510056
[ INFO] [1581480028.693493814]: Found parameter: svo/cam_d0, value: 0.932
[ WARN] [1581480028.696580355]: Cannot find value for parameter: svo/init_tz, assigning default: 0
[ WARN] [1581480028.697244516]: Cannot find value for parameter: svo/init_ty, assigning default: 0
[ WARN] [1581480028.698166501]: Cannot find value for parameter: svo/init_tx, assigning default: 0
[ INFO] [1581480028.698729852]: Found parameter: svo/init_rz, value: 0
[ INFO] [1581480028.702915812]: Found parameter: svo/init_ry, value: 0
[ INFO] [1581480028.704740701]: Found parameter: svo/init_rx, value: 3.14
[ WARN] [1581480028.705489481]: Cannot find value for parameter: svo/trace_name, assigning default: svo
[ WARN] [1581480028.706239823]: Cannot find value for parameter: svo/trace_dir, assigning default: /tmp
[ WARN] [1581480028.707494366]: Cannot find value for parameter: svo/n_pyr_levels, assigning default: 3
[ WARN] [1581480028.709314035]: Cannot find value for parameter: svo/use_imu, assigning default: 0
[... 59 lines cut ...]
[ INFO] [1581480074.451498691]: Init: KLT tracked 147 features
[ INFO] [1581480074.452042772]: Init: KLT 35.0074px average disparity.
[ INFO] [1581480074.500174268]: Init: KLT tracked 147 features
[ INFO] [1581480074.504019569]: Init: KLT 37.212px average disparity.
[ INFO] [1581480074.527515220]: Init: KLT tracked 147 features
[ INFO] [1581480074.528142452]: Init: KLT 39.0935px average disparity.
[ INFO] [1581480074.563754595]: Init: KLT tracked 147 features
[ INFO] [1581480074.565453457]: Init: KLT 41.0359px average disparity.
[ INFO] [1581480074.597098261]: Init: KLT tracked 147 features
[ INFO] [1581480074.598865943]: Init: KLT 43.2432px average disparity.
[ INFO] [1581480074.627300825]: Init: KLT tracked 147 features
[ INFO] [1581480074.629795131]: Init: KLT 45.2503px average disparity.
[ INFO] [1581480074.663960123]: Init: KLT tracked 146 features
[ INFO] [1581480074.664384233]: Init: KLT 47.631px average disparity.
[ INFO] [1581480074.706929732]: Init: KLT tracked 145 features
[ INFO] [1581480074.710322459]: Init: KLT 49.2348px average disparity.
[ INFO] [1581480074.739384205]: Init: KLT tracked 145 features
[ INFO] [1581480074.739804149]: Init: KLT 50.8822px average disparity.
[ INFO] [1581480074.752684038]: Init: Homography RANSAC 141 inliers.
[svo-1] process has died [pid 2709, exit code -11, cmd /home/chinthaka/catkin_ws/devel/lib/svo_ros/vo __name:=svo __log:=/home/chinthaka/.ros/log/06d84934-4d4c-11ea-8360-08002776c300/svo-1.log].
log file: /home/chinthaka/.ros/log/06d84934-4d4c-11ea-8360-08002776c300/svo-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

faustoTapia, 2020-06-26:
Any updates on this problem. I get exactly the same error.

zanazakaryaie, 2020-07-04:
I also get the same error (ROS Melodic)

zanazakaryaie, 2020-07-04:
Found the solution. Remove `-march=native` in all CMakeLists.txt files!

chinthaA, 2020-07-09:
Yes -march=native works. Thanks a ton.

closed 2020-07-09 by chinthaA

zxqhdu, 2025-01-16:
@zanazakaryaie it  has sloved my the same problem