uzh-rpg/rpg_svo #263: Jetson TX2 (cpu: aarch64)

zinuok, 2020-02-14:
I'm trying svo using 'Jetson TX2' on my drone.
The TX2 use cpu of 'aarch64', but svo doesn't support this, following errors occur:
c++: error: unrecognized command line option ‘-mmmx’

How can I fix it?

sachinkm308, 2020-04-11:
https://stackoverflow.com/questions/32523589/svo-fast-coner-detector-installation-plain-cmake-no-ros

Try this. I got it working. But stuck here

https://github.com/uzh-rpg/rpg_svo/issues/218

MRinel, 2020-06-13:
@zinuok have you test the power consumption?

AwosomeTony, 2021-01-27:
> https://stackoverflow.com/questions/32523589/svo-fast-coner-detector-installation-plain-cmake-no-ros
> 
> Try this. I got it working. But stuck here
> 
> #218

Excuse me, did you finish compiling the whole project? I meet the error when compiling faster_conor_10_sse.cpp
error:#error "This file requires SSE2 support. Check your compiler flags."