uzh-rpg/rpg_svo #264: Run SVO with ROS Melodic

geothan, 2020-03-18:
I followed your guidelines for setting up SVO with ROS:
https://github.com/uzh-rpg/rpg_svo/wiki/Installation:-ROS
i installed the last version of opencv and executed a successful catkin_make.

My problem is that the command: roslaunch svo_ros test_rig3.launch returns the below log:

user@ubuntu:~/catkin_ws$ roslaunch svo_ros test_rig3.launch
... logging to /home/user/.ros/log/f959d20c-695a-11ea-ac11-4074e0f52986/roslaunch-ubuntu-6059.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:40389/

SUMMARY
========

CLEAR PARAMETERS
 * /svo/

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /svo/cam_cx: 0.45905
 * /svo/cam_cy: 0.510056
 * /svo/cam_d0: 0.932
 * /svo/cam_fx: 0.509326
 * /svo/cam_fy: 0.796651
 * /svo/cam_height: 480
 * /svo/cam_model: ATAN
 * /svo/cam_topic: /camera/image_raw
 * /svo/cam_width: 752
 * /svo/grid_size: 30
 * /svo/init_rx: 3.14
 * /svo/init_ry: 0.0
 * /svo/init_rz: 0.0
 * /svo/loba_num_iter: 0
 * /svo/max_n_kfs: 10

NODES
  /
    svo (svo_ros/vo)

ROS_MASTER_URI=http://localhost:11311

process[svo-1]: started with pid [6075]
double free or corruption (out)
[svo-1] process has died [pid 6075, exit code -6, cmd /home/user/catkin_ws/devel/lib/svo_ros/vo __name:=svo __log:=/home/user/.ros/log/f959d20c-695a-11ea-ac11-4074e0f52986/svo-1.log].
log file: /home/user/.ros/log/f959d20c-695a-11ea-ac11-4074e0f52986/svo-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

 Is there any solution for that problem?

gustavojcorrea, 2020-05-18:
Any luck with this issue? I am getting the same thing.

zanazakaryaie, 2020-07-04:
Remove `-march=native` in all CMakeLists.txt files!

suntttabc, 2022-03-22:
Can you teach me how to Remove -march=native in all CMakeLists.txt files？Can you tell me the specific folder location，thanks a ton。