uzh-rpg/rpg_svo #265: live camera for NON-ROS build test_pipeline core dump error

Meda676, 2020-05-30:
I want to compile and execute the test_pipeline script for use SVO without ROS. My purpose is to change the runFromFolder function to read video from live stream. At the moment my problem is that, without changing anything in the test_pipeline.cpp script, after build it when I try to execute it I have a core dump error generated probabily by vikit's pinhole camera file..  Anyone know this issue?
Any help would be appreciated thanks!

tianfenhua123, 2020-06-16:
Hi, I also want to get reading video from live stream working. Is it possible to share your code?

> I want to compile and execute the test_pipeline script for use SVO without ROS. My purpose is to change the runFromFolder function to read video from live stream. At the moment my problem is that, [...]

Meda676, 2020-06-16:
To do that I used the code runFromCamera() from here: [https://github.com/uzh-rpg/rpg_svo/issues/18](url)

gauti1311, 2020-09-08:
refer the issue #202 .