uzh-rpg/rpg_svo #270: rpg_svo fatal error: sophus/se3.h: No such file or directory

Kalexo, 2021-03-17:
Hello,
I'm facing an error when I run catkin_make to use rpg_svo. Indeed, there is no header named sophus/se3.h or at least I didn't find one. I get the following errors:

In file included from /home/chappat/catkin_ws/src/rpg_vikit/vikit_common/src/math_utils.cpp:8:0:
/home/chappat/catkin_ws/src/rpg_vikit/vikit_common/include/vikit/math_utils.h:14:10: fatal error: sophus/se3.h: No such file or directory
 #include <sophus/se3.h>
          ^~~~~~~~~~~~~~
compilation terminated.
rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/build.make:110: recipe for target 'rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/math_utils.cpp.o' failed
make[2]: *** [rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/math_utils.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/chappat/catkin_ws/src/rpg_vikit/vikit_common/include/vikit/omni_camera.h:15:0,
                 from /home/chappat/catkin_ws/src/rpg_vikit/vikit_common/src/omni_camera.cpp:10:
/home/chappat/catkin_ws/src/rpg_vikit/vikit_common/include/vikit/math_utils.h:14:10: fatal error: sophus/se3.h: No such file or directory
 #include <sophus/se3.h>
          ^~~~~~~~~~~~~~
compilation terminated.
rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/build.make:86: recipe for target 'rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/omni_camera.cpp.o' failed
make[2]: *** [rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/omni_camera.cpp.o] Error 1
[ 40%] Built target imu_filter
[ 45%] Built target rviz_imu_plugin
[ 45%] Built target svo_msgs_generate_messages
In file included from /home/chappat/catkin_ws/src/rpg_vikit/vikit_common/include/vikit/atan_camera.h:20:0,
                 from /home/chappat/catkin_ws/src/rpg_vikit/vikit_common/src/atan_camera.cpp:14:
/home/chappat/catkin_ws/src/rpg_vikit/vikit_common/include/vikit/math_utils.h:14:10: fatal error: sophus/se3.h: No such file or directory
 #include <sophus/se3.h>

I am using the ROS Melodic version. Do you have any lead for this problem?
Thanks in advance

ferreram, 2021-03-29:
You either need to get the version of Sophus which was used by the time SVO was developed or change the headers (and more in the codes) to cope with the updates in Sophus.  Right now, sophus headers are ".hpp" files and not ".h" files anymore.

mext169, 2021-04-08:
I build svo without ROS and meet the same error like you.I change `vikit_common/CMakeLists.txt` as follow
`# FIND_PACKAGE(Sophus REQUIRED)
set(Sophus_INCLUDE_DIRS "/media/lab/m2_mext/Github_/rpg_svo_ws/Sophus")
set(Sophus_LIBS "/media/lab/m2_mext/Github_/rpg_svo_ws/Sophus/build/libSophus.so")`

`ADD_LIBRARY(${PROJECT_NAME} SHARED ${SOURCEFILES})
TARGET_LINK_LIBRARIES(${PROJECT_NAME}
  ${OpenCV_LIBS}
  ${Sophus_LIBS})`

my error was solved after that.