uzh-rpg/rpg_svo #276: Issue while running ./test_pipeline

srimannaini, 2022-06-16:
Hello,
I am trying to run the SVO without ROS, but I am having trouble while running it
Please, help me! 
[INFO] SVO initialized
reading image /home/sriman/svo/Datasets//sin2_tex2_h1_v8_d/img/frame_000002_0.png
[INFO] RESET
[INFO] Init: Selected first frame.
Frame-Id: 0 	#Features: 0 	Proc. Time: 4.252ms 
[INFO] Init: KLT tracked 565 features
[INFO] Init: KLT 25.3823px average disparity.
Frame-Id: 1 	#Features: 0 	Proc. Time: 2.34ms 
[INFO] Init: KLT tracked 543 features
[INFO] Init: KLT 50.8035px average disparity.
[INFO] Init: Homography RANSAC 543 inliers.
[INFO] Init: Selected second frame, triangulated initial map.
Frame-Id: 2 	#Features: 498 	Proc. Time: 2.637ms 
Frame-Id: 3 	#Features: 121 	Proc. Time: 1.5ms 
Frame-Id: 4 	#Features: 121 	Proc. Time: 0.691ms 
Frame-Id: 5 	#Features: 121 	Proc. Time: 0.698ms 
Frame-Id: 6 	#Features: 121 	Proc. Time: 0.704ms 
Frame-Id: 7 	#Features: 121 	Proc. Time: 0.695ms 
Frame-Id: 8 	#Features: 121 	Proc. Time: 0.694ms 
Frame-Id: 9 	#Features: 121 	Proc. Time: 0.673ms 
Frame-Id: 10 	#Features: 121 	Proc. Time: 0.768ms 
Frame-Id: 11 	#Features: 121 	Proc. Time: 0.715ms 
Frame-Id: 12 	#Features: 121 	Proc. Time: 0.8ms 
Frame-Id: 13 	#Features: 121 	Proc. Time: 0.74ms 
Frame-Id: 14 	#Features: 121 	Proc. Time: 0.84ms 
Frame-Id: 15 	#Features: 121 	Proc. Time: 0.775ms 
Frame-Id: 16 	#Features: 121 	Proc. Time: 0.881ms 
Frame-Id: 17 	#Features: 121 	Proc. Time: 0.773ms 
Frame-Id: 18 	#Features: 121 	Proc. Time: 0.934ms 
Frame-Id: 19 	#Features: 121 	Proc. Time: 0.772ms 
Frame-Id: 20 	#Features: 121 	Proc. Time: 0.847ms 
Frame-Id: 21 	#Features: 121 	Proc. Time: 0.73ms 
Frame-Id: 22 	#Features: 121 	Proc. Time: 0.801ms 
Frame-Id: 23 	#Features: 121 	Proc. Time: 0.869ms 
Frame-Id: 24 	#Features: 121 	Proc. Time: 0.899ms 
Frame-Id: 25 	#Features: 121 	Proc. Time: 0.772ms 
Frame-Id: 26 	#Features: 121 	Proc. Time: 0.945ms 
Frame-Id: 27 	#Features: 121 	Proc. Time: 0.755ms 
Frame-Id: 28 	#Features: 121 	Proc. Time: 0.866ms 
Frame-Id: 29 	#Features: 121 	Proc. Time: 0.7ms 
Frame-Id: 30 	#Features: 121 	Proc. Time: 0.841ms 
Frame-Id: 31 	#Features: 121 	Proc. Time: 0.763ms 
Frame-Id: 32 	#Features: 121 	Proc. Time: 0.93ms 
Frame-Id: 33 	#Features: 121 	Proc. Time: 0.766ms 
Frame-Id: 34 	#Features: 121 	Proc. Time: 0.884ms 
Frame-Id: 35 	#Features: 121 	Proc. Time: 0.832ms 
Frame-Id: 36 	#Features: 121 	Proc. Time: 0.903ms 
Frame-Id: 37 	#Features: 121 	Proc. Time: 0.809ms 
Frame-Id: 38 	#Features: 121 	Proc. Time: 0.893ms 
Frame-Id: 39 	#Features: 121 	Proc. Time: 0.829ms 
Frame-Id: 40 	#Features: 121 	Proc. Time: 0.927ms 
Frame-Id: 41 	#Features: 121 	Proc. Time: 0.723ms 
Frame-Id: 42 	#Features: 121 	Proc. Time: 0.961ms 
Frame-Id: 43 	#Features: 121 	Proc. Time: 0.775ms 
Frame-Id: 44 	#Features: 121 	Proc. Time: 0.921ms 
Frame-Id: 45 	#Features: 121 	Proc. Time: 0.906ms 
Frame-Id: 46 	#Features: 121 	Proc. Time: 0.924ms 
Frame-Id: 47 	#Features: 121 	Proc. Time: 0.829ms 
Frame-Id: 48 	#Features: 121 	Proc. Time: 0.929ms 
Frame-Id: 49 	#Features: 121 	Proc. Time: 0.885ms 
Frame-Id: 50 	#Features: 121 	Proc. Time: 0.841ms 
Frame-Id: 51 	#Features: 121 	Proc. Time: 0.821ms 
Frame-Id: 52 	#Features: 121 	Proc. Time: 0.892ms 
Frame-Id: 53 	#Features: 121 	Proc. Time: 0.768ms 
Frame-Id: 54 	#Features: 121 	Proc. Time: 0.893ms 
Frame-Id: 55 	#Features: 121 	Proc. Time: 0.789ms 
Frame-Id: 56 	#Features: 121 	Proc. Time: 0.893ms 
[... 108 lines cut ...]
Frame-Id: 165 	#Features: 121 	Proc. Time: 0.797ms 
Frame-Id: 166 	#Features: 121 	Proc. Time: 0.863ms 
Frame-Id: 167 	#Features: 121 	Proc. Time: 0.785ms 
Frame-Id: 168 	#Features: 121 	Proc. Time: 0.869ms 
Frame-Id: 169 	#Features: 121 	Proc. Time: 0.781ms 
Frame-Id: 170 	#Features: 121 	Proc. Time: 0.889ms 
Frame-Id: 171 	#Features: 121 	Proc. Time: 0.834ms 
Frame-Id: 172 	#Features: 121 	Proc. Time: 1.021ms 
Frame-Id: 173 	#Features: 121 	Proc. Time: 0.788ms 
Frame-Id: 174 	#Features: 121 	Proc. Time: 1.057ms 
Frame-Id: 175 	#Features: 121 	Proc. Time: 0.807ms 
Frame-Id: 176 	#Features: 121 	Proc. Time: 0.944ms 
Frame-Id: 177 	#Features: 121 	Proc. Time: 0.852ms 
Frame-Id: 178 	#Features: 121 	Proc. Time: 0.902ms 
Frame-Id: 179 	#Features: 121 	Proc. Time: 0.849ms 
Frame-Id: 180 	#Features: 121 	Proc. Time: 0.891ms 
Frame-Id: 181 	#Features: 121 	Proc. Time: 0.817ms 
Frame-Id: 182 	#Features: 121 	Proc. Time: 0.89ms 
Frame-Id: 183 	#Features: 121 	Proc. Time: 0.794ms 
Frame-Id: 184 	#Features: 121 	Proc. Time: 1.002ms 
Frame-Id: 185 	#Features: 121 	Proc. Time: 0.753ms 
[INFO] DepthFilter stop thread invoked.
[INFO] DepthFilter interrupt and join thread... 
[INFO] DepthFilter destructed.
Segmentation fault (core dumped)

DANIELHARDLY, 2022-11-17:
I have the same issue.
someone who know please help.

[INFO] SVO initialized
reading image /home/hardly/Datasets/sin2_tex2_h1_v8_d/img/frame_000002_0.png
[INFO] RESET
[INFO] Init: Selected first frame.
Frame-Id: 0 	#Features: 0 	Proc. Time: 4.638ms 
[INFO] Init: KLT tracked 565 features
[INFO] Init: KLT 25.3823px average disparity.
Frame-Id: 1 	#Features: 0 	Proc. Time: 6.789ms 
[INFO] Init: KLT tracked 543 features
[INFO] Init: KLT 50.8035px average disparity.
[INFO] Init: Homography RANSAC 543 inliers.
[INFO] Init: Selected second frame, triangulated initial map.
Frame-Id: 2 	#Features: 498 	Proc. Time: 7.321ms 
Frame-Id: 3 	#Features: 121 	Proc. Time: 2.226ms 
Frame-Id: 4 	#Features: 121 	Proc. Time: 0.813ms 
Frame-Id: 5 	#Features: 121 	Proc. Time: 0.783ms 
Frame-Id: 6 	#Features: 121 	Proc. Time: 0.8ms 
Frame-Id: 7 	#Features: 121 	Proc. Time: 0.901ms 
Frame-Id: 8 	#Features: 121 	Proc. Time: 0.834ms 
Frame-Id: 9 	#Features: 121 	Proc. Time: 0.816ms 
Frame-Id: 10 	#Features: 121 	Proc. Time: 0.943ms 
Frame-Id: 11 	#Features: 121 	Proc. Time: 0.863ms 
Frame-Id: 12 	#Features: 121 	Proc. Time: 1.001ms 
Frame-Id: 13 	#Features: 121 	Proc. Time: 0.901ms 
Frame-Id: 14 	#Features: 121 	Proc. Time: 1.079ms 
Frame-Id: 15 	#Features: 121 	Proc. Time: 1.275ms 
Frame-Id: 16 	#Features: 121 	Proc. Time: 1.221ms 
Frame-Id: 17 	#Features: 121 	Proc. Time: 0.944ms 
Frame-Id: 18 	#Features: 121 	Proc. Time: 1.145ms 
Frame-Id: 19 	#Features: 121 	Proc. Time: 1.181ms 
Frame-Id: 20 	#Features: 121 	Proc. Time: 1.118ms 
Frame-Id: 21 	#Features: 121 	Proc. Time: 0.904ms 
Frame-Id: 22 	#Features: 121 	Proc. Time: 0.98ms 
Frame-Id: 23 	#Features: 121 	Proc. Time: 1.117ms 
Frame-Id: 24 	#Features: 121 	Proc. Time: 1.163ms 
Frame-Id: 25 	#Features: 121 	Proc. Time: 0.939ms 
Frame-Id: 26 	#Features: 121 	Proc. Time: 1.174ms 
Frame-Id: 27 	#Features: 121 	Proc. Time: 1.015ms 
Frame-Id: 28 	#Features: 121 	Proc. Time: 1.115ms 
Frame-Id: 29 	#Features: 121 	Proc. Time: 0.923ms 
Frame-Id: 30 	#Features: 121 	Proc. Time: 1.166ms 
Frame-Id: 31 	#Features: 121 	Proc. Time: 1.053ms 
Frame-Id: 32 	#Features: 121 	Proc. Time: 1.195ms 
Frame-Id: 33 	#Features: 121 	Proc. Time: 0.953ms 
Frame-Id: 34 	#Features: 121 	Proc. Time: 1.286ms 
Frame-Id: 35 	#Features: 121 	Proc. Time: 1.101ms 
Frame-Id: 36 	#Features: 121 	Proc. Time: 1.171ms 
Frame-Id: 37 	#Features: 121 	Proc. Time: 1.014ms 
Frame-Id: 38 	#Features: 121 	Proc. Time: 1.236ms 
Frame-Id: 39 	#Features: 121 	Proc. Time: 1.046ms 
Frame-Id: 40 	#Features: 121 	Proc. Time: 1.236ms 
Frame-Id: 41 	#Features: 121 	Proc. Time: 0.952ms 
Frame-Id: 42 	#Features: 121 	Proc. Time: 1.266ms 
Frame-Id: 43 	#Features: 121 	Proc. Time: 1.001ms 
Frame-Id: 44 	#Features: 121 	Proc. Time: 1.146ms 
Frame-Id: 45 	#Features: 121 	Proc. Time: 1.022ms 
Frame-Id: 46 	#Features: 121 	Proc. Time: 1.327ms 
Frame-Id: 47 	#Features: 121 	Proc. Time: 1.037ms 
Frame-Id: 48 	#Features: 121 	Proc. Time: 1.166ms 
Frame-Id: 49 	#Features: 121 	Proc. Time: 1.065ms 
Frame-Id: 50 	#Features: 121 	Proc. Time: 1.238ms 
Frame-Id: 51 	#Features: 121 	Proc. Time: 1.064ms 
Frame-Id: 52 	#Features: 121 	Proc. Time: 1.153ms 
Frame-Id: 53 	#Features: 121 	Proc. Time: 0.96ms 
Frame-Id: 54 	#Features: 121 	Proc. Time: 1.357ms 
Frame-Id: 55 	#Features: 121 	Proc. Time: 1.053ms 
Frame-Id: 56 	#Features: 121 	Proc. Time: 1.161ms 
[... 108 lines cut ...]
Frame-Id: 165 	#Features: 121 	Proc. Time: 1.055ms 
Frame-Id: 166 	#Features: 121 	Proc. Time: 1.619ms 
Frame-Id: 167 	#Features: 121 	Proc. Time: 1.094ms 
Frame-Id: 168 	#Features: 121 	Proc. Time: 1.257ms 
Frame-Id: 169 	#Features: 121 	Proc. Time: 1.045ms 
Frame-Id: 170 	#Features: 121 	Proc. Time: 1.453ms 
Frame-Id: 171 	#Features: 121 	Proc. Time: 1.163ms 
Frame-Id: 172 	#Features: 121 	Proc. Time: 1.293ms 
Frame-Id: 173 	#Features: 121 	Proc. Time: 0.971ms 
Frame-Id: 174 	#Features: 121 	Proc. Time: 1.588ms 
Frame-Id: 175 	#Features: 121 	Proc. Time: 1.169ms 
Frame-Id: 176 	#Features: 121 	Proc. Time: 1.285ms 
Frame-Id: 177 	#Features: 121 	Proc. Time: 1.096ms 
Frame-Id: 178 	#Features: 121 	Proc. Time: 1.433ms 
Frame-Id: 179 	#Features: 121 	Proc. Time: 1.205ms 
Frame-Id: 180 	#Features: 121 	Proc. Time: 1.194ms 
Frame-Id: 181 	#Features: 121 	Proc. Time: 1.415ms 
Frame-Id: 182 	#Features: 121 	Proc. Time: 1.367ms 
Frame-Id: 183 	#Features: 121 	Proc. Time: 1.111ms 
Frame-Id: 184 	#Features: 121 	Proc. Time: 1.229ms 
Frame-Id: 185 	#Features: 121 	Proc. Time: 1.24ms 
[INFO] DepthFilter stop thread invoked.
[INFO] DepthFilter interrupt and join thread... 
[INFO] DepthFilter destructed.
Segmentation fault (core dumped)

lsorgi, 2023-03-14:
has anyone a solution for this issue? is this repo being monitored/maintained at all?