uzh-rpg/rpg_svo #282: Compilation error (without ROS)

unicornuniform, 2023-08-28:
> am@14IAU7:~/lib/rpg_svo/svo/build$ make 
> [  3%] Building CXX object CMakeFiles/svo.dir/src/frame_handler_mono.cpp.o
> /home/sam/lib/rpg_svo/svo/src/frame_handler_mono.cpp: In member function ‘virtual void svo::FrameHandlerMono::initialize()’:
> /home/sam/lib/rpg_svo/svo/src/frame_handler_mono.cpp:48:72: error: ‘_1’ was not declared in this scope
>    48 |       &MapPointCandidates::newCandidatePoint, &map_.point_candidates_, _1, _2);
>       |                                                                        ^~
>

And:

> /home/sam/lib/rpg_svo/svo/src/frame_handler_mono.cpp:48:76: error: ‘_2’ was not declared in this scope
>    48 |       &MapPointCandidates::newCandidatePoint, &map_.point_candidates_, _1, _2);
>       |                                                                            ^~
>

catnuko, 2023-11-28:
same error

cchun319, 2024-02-29:
you can try replace it with `boost::placeholders::_1`